negativo17/nvidia-driver #171: Failed to initialize NVML: Driver/library version mismatch (RHEL8.10)

ElCoyote27, 2025-02-05:
Hi Simone,
With the latest rpms for RHEL8 things no longer work on RHEL8:

```
# rpm -qa|grep nvidia|sort
akmod-nvidia-570.86.16-1.el8.x86_64
kmod-nvidia-4.18.0-553.el8_10-570.86.16-1.el8.x86_64
libnvidia-cfg-570.86.16-1.el8.x86_64
libnvidia-fbc-570.86.16-1.el8.i686
libnvidia-fbc-570.86.16-1.el8.x86_64
libnvidia-ml-570.86.16-1.el8.i686
libnvidia-ml-570.86.16-1.el8.x86_64
nvidia-driver-570.86.16-1.el8.x86_64
nvidia-driver-cuda-570.86.16-1.el8.x86_64
nvidia-driver-cuda-libs-570.86.16-1.el8.i686
nvidia-driver-cuda-libs-570.86.16-1.el8.x86_64
nvidia-driver-libs-570.86.16-1.el8.i686
nvidia-driver-libs-570.86.16-1.el8.x86_64
nvidia-kmod-common-570.86.16-1.el8.noarch
nvidia-libXNVCtrl-570.86.16-1.el8.x86_64
nvidia-libXNVCtrl-devel-570.86.16-1.el8.x86_64
nvidia-modprobe-570.86.16-1.el8.x86_64
nvidia-persistenced-570.86.16-1.el8.x86_64
nvidia-settings-570.86.16-1.el8.x86_64
nvidia-xconfig-570.86.16-1.el8.x86_64
pcp-pmda-nvidia-gpu-5.3.7-22.el8_10.x86_64
xorg-x11-nvidia-570.86.16-1.el8.x86_64

# nvidia-smi
Failed to initialize NVML: Driver/library version mismatch
NVML library version: 570.86
```

scaronni, 2025-02-05:
Sorry but not enough information.

It seems you're still running the old kernel. Are you on kernel `4.18.0-553.el8_10` as your output suggests?

```
$ uname -r
$ cat /proc/driver/nvidia/version
```

I assume this is a desktop, and you want to run CUDA applications. I don't know what's your use case, but you can remove some stuff and rebuild the `akmod`:

```
# rpm -e nvidia-libXNVCtrl-devel nvidia-xconfig nvidia-driver-cuda-libs.i686
# rpm -e kmod-nvidia-4.18.0-553.el8_10-570.86.16-1.el8.x86_64
# akmods -f
```

ElCoyote27, 2025-02-05:
Hi @scaronni 553 is the kABI family of the kernel I am running (8.10).

```
[root@daltigoth ~]# uname -r
4.18.0-553.36.1.el8_10.x86_64
[root@daltigoth ~]# cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX x86_64 Kernel Module  570.86.15  Thu Jan 23 23:23:10 UTC 2025
GCC version:  gcc version 8.5.0 20210514 (Red Hat 8.5.0-22) (GCC) 

```

Yes, I would like to run CUDA apps on this Desktop/Server as well.
I've already rebuilt the kmod a few times.

ElCoyote27, 2025-02-05:
Also, I wanted to keep both 32bit and 64bit libraries to support legacy apps.

ElCoyote27, 2025-02-05:
I removed all 32bit libs and did rebuild the kmod, same issue:

```
[root@daltigoth ~]# rpm -qa|grep nvidia|sort
akmod-nvidia-570.86.16-1.el8.x86_64
kmod-nvidia-4.18.0-553.el8_10-570.86.16-1.el8.x86_64
libnvidia-cfg-570.86.16-1.el8.x86_64
libnvidia-fbc-570.86.16-1.el8.x86_64
libnvidia-ml-570.86.16-1.el8.x86_64
nvidia-driver-570.86.16-1.el8.x86_64
nvidia-driver-cuda-570.86.16-1.el8.x86_64
nvidia-driver-cuda-libs-570.86.16-1.el8.x86_64
nvidia-driver-libs-570.86.16-1.el8.x86_64
nvidia-kmod-common-570.86.16-1.el8.noarch
nvidia-libXNVCtrl-570.86.16-1.el8.x86_64
nvidia-modprobe-570.86.16-1.el8.x86_64
nvidia-persistenced-570.86.16-1.el8.x86_64
nvidia-settings-570.86.16-1.el8.x86_64
nvidia-xconfig-570.86.16-1.el8.x86_64
pcp-pmda-nvidia-gpu-5.3.7-22.el8_10.x86_64
xorg-x11-nvidia-570.86.16-1.el8.x86_64
[root@daltigoth ~]# modinfo nvidia -F version
570.86.16
[root@daltigoth ~]# nvidia-smi 
Failed to initialize NVML: Driver/library version mismatch
NVML library version: 570.86

```

ElCoyote27, 2025-02-05:
Got it working on two other RHEL8 hosts so I guess everything is fine. Thank you.

closed 2025-02-05 by ElCoyote27

scaronni, 2025-02-05:
> Hi [@scaronni](https://github.com/scaronni) 553 is the kABI family of the kernel I am running (8.10).
>
> ```
> [root@daltigoth ~]# uname -r
> 4.18.0-553.36.1.el8_10.x86_64
> [root@daltigoth ~]# cat [...]

This shows that you are running the 570.86.16 driver but have the 570.86.15 kernel module loaded.

scaronni, 2025-02-05:
I think you just missed a reboot after rebuilding the module, as the old one was still loaded.

ElCoyote27, 2025-02-05:
Yes, I most likely missed that.. I thought I had checked, though. Sorry about the noise. Thank you once again for making these rpms available. :)