negativo17/nvidia-driver #174: Fails to build on F42 6.14.1 kernel

nerijus, 2025-04-11:
`/var/cache/akmods/nvidia/570.133.07-1-for-6.14.1-300.fc42.x86_64.failed.log` has:
```
2025/04/11 17:03:05 akmodsbuild: make[4]: *** [/usr/src/kernels/6.14.1-300.fc42.x86_64/scripts/Makefile.build:207: nvidia/nv-nano-timer.o] Error 1
2025/04/11 17:03:05 akmodsbuild: ././common/inc/nv-linux.h: In function ‘nv_phys_to_dma’:
2025/04/11 17:03:05 akmodsbuild: ././common/inc/nv-linux.h:709:12: error: implicit declaration of function ‘phys_to_dma’; did you mean ‘nv_phys_to_dma’? [-Wimplicit-function-declaration]

...

2025/04/11 17:03:05 akmodsbuild: In file included from /usr/src/kernels/6.14.1-300.fc42.x86_64/include/sound/hdaudio.h:18,
2025/04/11 17:03:05 akmodsbuild:                  from nvidia/nv.c:75:
2025/04/11 17:03:05 akmodsbuild: /usr/src/kernels/6.14.1-300.fc42.x86_64/include/sound/pcm.h: In function ‘snd_pcm_gettime’:
2025/04/11 17:03:05 akmodsbuild: /usr/src/kernels/6.14.1-300.fc42.x86_64/include/sound/pcm.h:1292:32: error: passing argument 1 of ‘ktime_get_ts64’ from incompatible pointer type [
-Wincompatible-pointer-types]
2025/04/11 17:03:05 akmodsbuild:  1292 |                 ktime_get_ts64(tv);
2025/04/11 17:03:05 akmodsbuild:       |                                ^~
2025/04/11 17:03:05 akmodsbuild:       |                                |
2025/04/11 17:03:05 akmodsbuild:       |                                struct timespec64 *
2025/04/11 17:03:05 akmodsbuild: /usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/timekeeping.h:43:47: note: expected ‘struct timespec64 *’ but argument is of type ‘struct tim
espec64 *’
2025/04/11 17:03:05 akmodsbuild:    43 | extern void ktime_get_ts64(struct timespec64 *ts);
2025/04/11 17:03:05 akmodsbuild:       |                            ~~~~~~~~~~~~~~~~~~~^~
2025/04/11 17:03:05 akmodsbuild: /usr/src/kernels/6.14.1-300.fc42.x86_64/include/sound/pcm.h:1295:36: error: passing argument 1 of ‘ktime_get_raw_ts64’ from incompatible pointer ty
pe [-Wincompatible-pointer-types]
2025/04/11 17:03:05 akmodsbuild:  1295 |                 ktime_get_raw_ts64(tv);
2025/04/11 17:03:05 akmodsbuild:       |                                    ^~
2025/04/11 17:03:05 akmodsbuild:       |                                    |
2025/04/11 17:03:05 akmodsbuild:       |                                    struct timespec64 *
2025/04/11 17:03:05 akmodsbuild: /usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/timekeeping.h:42:51: note: expected ‘struct timespec64 *’ but argument is of type ‘struct tim
espec64 *’
```
I will try to attach the log file.

nerijus, 2025-04-11:
[570.133.07-1-for-6.14.1-300.fc42.x86_64.failed.log](https://github.com/user-attachments/files/19707706/570.133.07-1-for-6.14.1-300.fc42.x86_64.failed.log)

scaronni, 2025-04-12:
Coming online shortly with fixes also for above 6.14.1, sorry for the delay.

closed 2025-04-12 by scaronni