negativo17/nvidia-driver #175: Installing CUDA only driver for secondary Nvidia GPU forces X11 mode on primary AMD GPU (i.e. breaks Wayland)

isloven, 2025-05-05:
Test setup: Fedora 42 box with an AMD GPU used as the primary monitor output with a secondary GPU used for CUDA only. Using a non Nvidia GPU for desktop is a common setup for number crunching workstations to prevent laggy screen and to be able to work while CUDA stuff is running in the background.

Without Nvidia driver installed, the AMD GPU desktop shows Wayland as the backend and works without issues.

I followed the instructions [here](https://negativo17.org/nvidia-driver/#CUDA_installations) to install the CUDA only driver:

`# dnf install nvidia-driver-cuda cuda-devel`

After a reboot, the CUDA stuff is working (tested with PyTorch) but the desktop switches to Wayland mode.

The only suspect in the kernel log is this:

> May 05 14:44:42 box gnome-shell[3145]: Running GNOME Shell (using mutter 48.2) as a Wayland display server
> May 05 14:44:42 box gnome-shell[3145]: Enabling experimental feature 'scale-monitor-framebuffer'
> May 05 14:44:42 box gnome-shell[3145]: Enabling experimental feature 'xwayland-native-scaling'
> May 05 14:44:43 box rtkit-daemon[2677]: Successfully made thread 3163 of process 3145 (/usr/bin/gnome-shell) owned by '42' high priority at nice level -15.
> May 05 14:44:43 box gnome-shell[3145]: Thread 'KMS thread' will be using high priority scheduling
> May 05 14:44:44 box org.gnome.Shell.desktop[3145]: libEGL warning: egl: failed to create dri2 screen
> May 05 14:44:44 box org.gnome.Shell.desktop[3145]: libEGL warning: egl: failed to create dri2 screen
> May 05 14:44:44 box audit[3145]: ANOM_ABEND auid=4294967295 uid=42 gid=42 ses=4294967295 subj=system_u:system_r:xdm_t:s0-s0:c0.c1023 pid=3145 comm="gnome-shell" exe="/usr/bin/gnome-shell" sig=11 res=1
> May 05 14:44:44 box kernel: show_signal_msg: 48 callbacks suppressed
> May 05 14:44:44 box kernel: gnome-shell[3145]: segfault at 8 ip 00007f774522d14e sp 00007ffed683f680 error 4 in libgallium-25.0.4.so[2d14e,7f7745200000+1b1c000] likely on CPU 9 (core 9, socket 0)
> May 05 14:44:44 box kernel: Code: 83 ec 18 48 8b 57 58 0f b6 05 8f 6c ba 02 84 c0 0f 84 8e 00 00 00 0f b6 05 7f 6c ba 02 4c 8d 2d 98 a2 a6 02 88 83 68 01 00 00 <83> 7a 08 03 7e 17 4c 8d 2d 65 a2 a6 02 48 83 7a 38 00 48 8d 05 79
> May 05 14:44:44 box systemd-coredump[3232]: Process 3145 (gnome-shell) of user 42 terminated abnormally with signal 11/SEGV, processing...
> May 05 14:44:44 box systemd[1]: Created slice system-systemd\x2dcoredump.slice - Slice /system/systemd-coredump.
> May 05 14:44:44 box audit: BPF prog-id=63 op=LOAD
> May 05 14:44:44 box audit: BPF prog-id=64 op=LOAD
> May 05 14:44:44 box audit: BPF prog-id=65 op=LOAD
> May 05 14:44:44 box systemd[1]: Started systemd-coredump@0-3232-0.service - Process Core Dump (PID 3232/UID 0).
> May 05 14:44:44 box audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@0-3232-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
> May 05 14:44:45 box systemd-coredump[3233]: [🡕] Process 3145 (gnome-shell) of user 42 dumped core.
> Module /usr/bin/gnome-shell from rpm gnome-shell-48.1-1.fc42.x86_64
> Module dri_gbm.so from rpm mesa-25.0.4-2.fc42.x86_64
> Module libpciaccess.so.0 from rpm libpciaccess-0.16-15.fc42.x86_64
> Module libedit.so.0 from rpm libedit-3.1-55.20250104cvs.fc42.x86_64
> Module libdrm_intel.so.1 from rpm libdrm-2.4.124-2.fc42.x86_64
> Module libdrm_amdgpu.so.1 from rpm libdrm-2.4.124-2.fc42.x86_64
> Module libelf.so.1 from rpm elfutils-0.193-2.fc42.x86_64
> Module libxshmfence.so.1 from rpm libxshmfence-1.3.2-6.fc42.x86_64
> Module libxcb-sync.so.1 from rpm libxcb-1.17.0-5.fc42.x86_64
> Module libsensors.so.4 from rpm lm_sensors-3.6.0-22.fc42.x86_64
> Module libSPIRV-Tools.so from rpm spirv-tools-2025.1-1.fc42.x86_64
> Module libzstd.so.1 from rpm zstd-1.5.7-1.fc42.x86_64
> Module libxcb-present.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
> Module libxcb-dri3.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
> Module libwayland-client.so.0 from rpm wayland-1.23.1-1.fc42.x86_64
> Module libxcb-xfixes.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
> Module libgallium-25.0.4.so from rpm mesa-25.0.4-2.fc42.x86_64
> Module libEGL_mesa.so.0 from rpm mesa-25.0.4-2.fc42.x86_64

...
                                             
> Stack trace of thread 3145:
> #0  0x00007f774522d14e drisw_init_screen (libgallium-25.0.4.so + 0x2d14e)
> #1  0x00007f7745228035 driCreateNewScreen3 (libgallium-25.0.4.so + 0x28035)
> #2  0x00007f7768184ed2 dri2_create_screen (libEGL_mesa.so.0 + 0x16ed2)
> #3  0x00007f7768186775 dri2_initialize_device (libEGL_mesa.so.0 + 0x18775)
> #4  0x00007f7768185788 dri2_initialize (libEGL_mesa.so.0 + 0x17788)
> #5  0x00007f7768173f5e eglInitialize (libEGL_mesa.so.0 + 0x5f5e)
> #6  0x00007f777190dadd meta_render_device_egl_stream_create_egl_display (libmutter-16.so.0 + 0x10dadd)
> #7  0x00007f777198b6ae meta_render_device_initable_init (libmutter-16.so.0 + 0x18b6ae)
> #8  0x00007f777190d90e meta_render_device_egl_stream_initable_init (libmutter-16.so.0 + 0x10d90e)
> #9  0x00007f777207a13c g_initable_new_valist (libgio-2.0.so.0 + 0x4913c)
> #10 0x00007f777207a22b g_initable_new (libgio-2.0.so.0 + 0x4922b)
> #11 0x00007f7771966092 create_render_device.lto_priv.0 (libmutter-16.so.0 + 0x166092)
> #12 0x00007f777196636e add_drm_device (libmutter-16.so.0 + 0x16636e)
> #13 0x00007f7771966aa6 meta_backend_native_init_basic (libmutter-16.so.0 + 0x166aa6)
> #14 0x00007f7771845937 meta_backend_initable_init (libmutter-16.so.0 + 0x45937)
> #15 0x00007f777207a13c g_initable_new_valist (libgio-2.0.so.0 + 0x4913c)
> #16 0x00007f777207a22b g_initable_new (libgio-2.0.so.0 + 0x4922b)
> #17 0x00007f77718a119e meta_context_real_setup (libmutter-16.so.0 + 0xa119e)
> #18 0x00007f77718a6992 meta_context_main_setup (libmutter-16.so.0 + 0xa6992)
> #19 0x00007f77718a6f3d meta_context_setup (libmutter-16.so.0 + 0xa6f3d)
> #20 0x000055b0b1fb1074 main (/usr/bin/gnome-shell + 0x1074)
> #21 0x00007f77716115f5 __libc_start_call_main (libc.so.6 + 0x35f5)
> #22 0x00007f77716116a8 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x36a8)
> #23 0x000055b0b1fb1745 _start (/usr/bin/gnome-shell + 0x1745)

...

Is there a recommended procedure for setting up CUDA only driver for secondary Nvidia GPU and/or is this a bug?

Thanks!

scaronni, 2025-05-07:
```rpm -qa \*nvidia\* | sort```

isloven, 2025-05-09:
`$ rpm -qa \*nvidia\* | sort`

> akmod-nvidia-570.144-1.fc42.x86_64
> kmod-nvidia-6.14.4-300.fc42.x86_64-570.144-1.fc42.x86_64
> libnvidia-cfg-570.144-1.fc42.x86_64
> libnvidia-ml-570.144-1.fc42.x86_64
> nvidia-driver-cuda-570.144-1.fc42.x86_64
> nvidia-driver-cuda-libs-570.144-1.fc42.x86_64
> nvidia-gpu-firmware-20250410-1.fc42.noarch
> nvidia-kmod-common-570.144-1.fc42.noarch
> nvidia-modprobe-570.144-1.fc42.x86_64
> nvidia-persistenced-570.144-1.fc42.x86_64
> xorg-x11-drv-nvidia-kmodsrc-570.144-1.fc42.x86_64

isloven, 2025-05-09:
Not sure if this is the same issue, but it seems to have similar symptoms:
https://bugs.launchpad.net/ubuntu/+source/mesa/+bug/2101817

It links to these freedesktop.org issues which seem to be related:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/12678
https://gitlab.freedesktop.org/mesa/mesa/-/issues/12300
https://gitlab.freedesktop.org/mesa/mesa/-/issues/12462

This seems to be the (work in progress) fix?
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/34380

So this might not have anything to do with your packages of the nvidia driver. 

I'd be happy to try anything you may recommend however if it helps troubleshoot this issue. Thanks.

scaronni, 2025-05-12:
You're mixing packages from my repository and RPMFusion, that is not supported. Please use one or the other.

scaronni, 2025-05-12:
> Without Nvidia driver installed, the AMD GPU desktop shows Wayland as the backend and works without issues.
> 
> After a reboot, the CUDA stuff is working (tested with PyTorch) but the desktop switches to Wayland mode.

I'm not following here, you said it's on Wayland and then that switches to Wayland mode?

isloven, 2025-05-12:
> You're mixing packages from my repository and RPMFusion, that is not supported. Please use one or the other.

I went back and removed all the nvidia packages:
```
$ rpm -qa \*nvidia\* | sort
$
```

and then again followed the instructions on https://negativo17.org/nvidia-driver and it seems like it automatically installs stuff from rpmfusion-nonfree-updates repository:

```
$ sudo dnf install nvidia-driver-cuda
Updating and loading repositories:
Repositories loaded.
Package                       Arch   Version                       Repository                     Size
Installing:
 nvidia-driver-cuda           x86_64 3:570.144-1.fc42              fedora-nvidia               1.4 MiB
Installing dependencies:
 akmod-nvidia                 x86_64 3:570.144-1.fc42              rpmfusion-nonfree-updates  92.2 KiB
 akmods                       noarch 0.6.0-9.fc42                  fedora                     64.2 KiB
 kernel-devel-matched         x86_64 6.14.5-300.fc42               updates                     0.0   B
 kmodtool                     noarch 1.1-13.fc42                   fedora                     27.6 KiB
 libnvidia-cfg                x86_64 3:570.144-1.fc42              fedora-nvidia             394.0 KiB
 libnvidia-ml                 x86_64 3:570.144-1.fc42              fedora-nvidia               2.1 MiB
 nvidia-driver-cuda-libs      x86_64 3:570.144-1.fc42              fedora-nvidia             271.4 MiB
 nvidia-kmod-common           noarch 3:570.144-1.fc42              fedora-nvidia              87.9 MiB
 nvidia-modprobe              x86_64 3:570.144-1.fc42              rpmfusion-nonfree-updates  50.9 KiB
 nvidia-persistenced          x86_64 3:570.144-1.fc42              rpmfusion-nonfree-updates  54.4 KiB
 openssl                      x86_64 1:3.2.4-3.fc42                fedora                      1.7 MiB
 xorg-x11-drv-nvidia-kmodsrc  x86_64 3:570.144-1.fc42              rpmfusion-nonfree-updates  76.0 MiB

Transaction Summary:
 Installing:        13 packages

Total size of inbound packages is 188 MiB. Need to download 188 MiB.
After this operation, 441 MiB extra will be used (install 441 MiB, remove 0 B).
Is this ok [y/N]:
```

Am I supposed to disable rpmfusion repository to use yours?

isloven, 2025-05-12:
> I'm not following here, you said it's on Wayland and then that switches to Wayland mode?

The desktop works on Wayland until I install Nvidia driver. Once I install Nvidia driver, on the next boot it uses X11 (I have no option to use Wayland in GDM).

1) Install Fedora on a box that has monitor plugged into a radeon card and nothing plugged into an nvidia card
2) Boot up into desktop - it uses wayland
3) sudo dnf install nvidia-driver-cuda
4) Reboot
5) Desktop now uses X11 and wayland doesn't seem to be an option

closed 2025-11-02 by isloven