negativo17/nvidia-driver #176: [feature] Set vendor string in RPMs

pointydev, 2025-05-07:
Hey there,

I think it'd be useful to have a proper vendor string set in package specs (currently it reads `<NULL>`) so that users can utilise `allow_vendor_change=False` in order to properly prevent packages from other repos that may conflict from being installed.

While I'm here, I'd also just like to bump our email conversation about rawhide repos from a few weeks ago. 😅

Thanks,
Elliott

scaronni, 2025-05-08:
I've added the Vendor and Packager RPM tags, but they are generated at build time like Fedora does, as those are explicitly forbidden by the package guidelines. This means packages will eventually get that over time.

Regarding rawhide, I was hoping to land some more initrd stuff in the packages before proceeding, but I'll save those for later. Building now.

scaronni, 2025-05-08:
I've built only the driver packages for now, the rest of the packages would be the same as fedora 42.

scaronni, 2025-05-08:
```
$ rpm -qpi nvidia-driver-575.51.02-1.fc43.x86_64.rpm | grep -iE "packager|vendor"
Packager    : negativo17.org
Vendor      : negativo17.org
```

closed 2025-05-08 by scaronni