negativo17/nvidia-driver #177: 575.57.08 not available for install

drelbszoomer, 2025-05-30:
This is maybe a stupid question, but I notice that I cannot see any of the "new feature branch release" NVIDIA driver versions (the 575 series) at all when I query this repo with DNF. 

Just the 570.xx versions.

Is that intentional or am I missing something? How can I install the 575 series?

```
dnf search --showduplicates nvidia-driver                                               
                                                                                                         
Updating and loading repositories:
Repositories loaded.
Matched fields: name (exact)
 nvidia-driver-3:570.144-1.fc42.x86_64: NVIDIA's proprietary display driver for NVIDIA graphic cards
 nvidia-driver-3:570.153.02-1.fc42.x86_64: NVIDIA's proprietary display driver for NVIDIA graphic cards
Matched fields: name, summary
 nvidia-driver-cuda-3:570.144-1.fc42.x86_64: CUDA integration for nvidia-driver
 nvidia-driver-cuda-3:570.153.02-1.fc42.x86_64: CUDA integration for nvidia-driver
 nvidia-driver-cuda-libs-3:570.144-1.fc42.i686: Libraries for nvidia-driver-cuda
 nvidia-driver-cuda-libs-3:570.144-1.fc42.x86_64: Libraries for nvidia-driver-cuda
 nvidia-driver-cuda-libs-3:570.153.02-1.fc42.i686: Libraries for nvidia-driver-cuda
 nvidia-driver-cuda-libs-3:570.153.02-1.fc42.x86_64: Libraries for nvidia-driver-cuda
 nvidia-driver-libs-3:570.144-1.fc42.i686: Libraries for nvidia-driver
 nvidia-driver-libs-3:570.144-1.fc42.x86_64: Libraries for nvidia-driver
 nvidia-driver-libs-3:570.153.02-1.fc42.i686: Libraries for nvidia-driver
 nvidia-driver-libs-3:570.153.02-1.fc42.x86_64: Libraries for nvidia-driver
Matched fields: name
 libva-nvidia-driver-0:0.0.13^20241210git9c4a3c7-2.fc42.i686: A VA-API implemention using NVIDIA's NVDEC
 libva-nvidia-driver-0:0.0.13^20241210git9c4a3c7-2.fc42.x86_64: A VA-API implemention using NVIDIA's NVDEC
 libva-nvidia-driver-0:0.0.13^20250419gitc2860cc-1.fc42.i686: A VA-API implemention using NVIDIA's NVDEC
 libva-nvidia-driver-0:0.0.13^20250419gitc2860cc-1.fc42.x86_64: A VA-API implemention using NVIDIA's NVDEC
 libva-nvidia-driver-1:0.0.13^20250118gitc519e97-2.fc42.i686: VA-API user mode driver for Nvidia GPUs
 libva-nvidia-driver-1:0.0.13^20250118gitc519e97-2.fc42.x86_64: VA-API user mode driver for Nvidia GPUs
```

scaronni, 2025-05-31:
It's coming today.

closed 2025-05-31 by scaronni

scaronni, 2025-06-02:
Well, got distracted by the weekend. Actually *today*.